RetroDriven/Pocket_Updater #10: Error retrieving Drive Information

esmith13, 2022-09-20:
Pocket Updater v1.1.0
Upon clicking "Update Pocket" I am immediately greeted with and "Error!" window stating "Error retrieving Drive Information". 
Clicking "OK" brings me to the "Update Location:" setting.
"Current DIrectory" works as expected but "Removable Storage" has no options to choose from. Clicking the green refresh button immediately pops up the original error message again.  This happens 100% of the time, regardless of whether or not removable storage is inserted in my PC via USB, built-in SD reader or not at all.

The PC I am using is a WIndows 10 Professional desktop on a corporate domain. I am the admin of both the PC and the domain so it is not an account rights issue. The PC has 13 drive letters in use (not counting those used by removable drives). 11 letters are assigned to local hard disks and two are mapped network drives. Removable media like SD cards or thumb drives when inserted generate additional drive letters.

Other software (both commercial and not) that specifically tries to point you to or limit you to removable media works as expected. Examples are SD Card Formatter, Etcher, Ultimaker Cura (auto redirect to devices detected as removable media when inserted), and a couple .NET apps that are specifically for manipulating SD card data only and completely ignore non-removable disks.

Sorry for the excessive info, wasn't sure if any of that would make it easier to track down the problem or not.  😄 

Thanks so much for the awesome app!

NeitherMeNorYou, 2022-09-20:
I have the same issue on my PC, but using Windows 11 Pro. My drive setup is 3 internal drives, 5 SMB network shares, 1 DVD drive, multiple removable drives that generally aren't connected, and the Pocket MicroSD is connected to my PC via a Kingston Multi cardreader, also only occasionally connected.

I've tried two slots on Kingston card reader, too. Directly in the MicroSD slot and in the SD slot via adapter, I get the same error message for both.

Updating everything locally to where the exe is located works perfectly fine.

RetroDriven, 2022-09-23:
Someone just submitted a PR to fix this. I'll create a test build for you both to test and confirm that it works.

RetroDriven, 2022-09-23:
@NeitherMeNorYou @esmith13 Please try the Test build below to see if the PR fixes the issue and let me know when possile.

[https://gofile.io/d/oZchfi](url)

NeitherMeNorYou, 2022-09-24:
@RetroDriven I was able to successfully update the cores on my setup with the test file, it found the MicroSD without any issues.

RetroDriven, 2022-09-24:
> @RetroDriven I was able to successfully update the cores on my setup with the test file, it found the MicroSD without any issues.

Excellent! Thanks for the quick test and feedback.

@esmith13 Can you test and confirm this as well please?

esmith13, 2022-09-24:
@RetroDriven confirmed working as expected for me as well! Thanks so much!

RetroDriven, 2022-09-24:
> @RetroDriven confirmed working as expected for me as well! Thanks so much!

Perfect! @LoneBoco boot me to fixing this issue so all the credit goes there.

closed 2022-09-24 by RetroDriven