RetroDriven/Pocket_Updater #14: Use an external API for getting the up-to-date list of cores

goronfreeman, 2022-09-22:
This is a cross-post of an issue I created in mattpannella/pocket_core_autoupdate_net#16. I noticed you use the core list in that repo already and would love your input on this.

> I've noticed there are several tools in the community for updating Pocket cores. I have contributed to the [`update-pocket`](https://github.com/rivergillis/update-pocket) project, I'm aware of your project, and I know there is now a Windows application. Every time a new core is released, every project needs to update its own internal list of cores. It is my proposal that instead, we create a new API outside of these projects where the core list can be retrieved from.
> 
> We already have the great resource [`openfpga-cores-inventory`](https://github.com/joshcampbell191/openfpga-cores-inventory), so I suggest that we utilize that for hosting the API. This is another place that has to be independently updated every time a core is released, and is the best resource for members of the community to find the list of available cores. I already have the API put together in a local branch, ready to be put up as a PR, but for this to be successful, the developers of updater tools should all agree on a set JSON structure for the data.
> 
> I would love to get your feedback on this, and if there is a consensus in the community that this is a good idea, I'm more than happy to stand up the API myself.

goronfreeman, 2022-09-22:
I added an issue to keep track of the discussion here: https://github.com/joshcampbell191/openfpga-cores-inventory#5

RetroDriven, 2022-09-23:
I'd love to see what you have and how it would work for Matt and I.

goronfreeman, 2022-09-24:
> I'd love to see what you have and how it would work for Matt and I.

Feel free to join in the coversation at joshcampbell191/openfpga-cores-inventory#6. @mattpannella provided us some information, but I'd love your input as well. The updated `README` has the documented API. It was closely based off of what you are already using in Matt's project.

RetroDriven, 2022-10-02:
This has been implemented in v1.3.1 #19

closed 2022-10-02 by RetroDriven