RetroDriven/Pocket_Updater #38: Scaling Issue with 4K display on Windows 11

Wesleyrpg, 2022-11-16:
When using this app on Windows 11 with 250% scaling, it almost becomes unusable.

There is definitely something funky going on with how the app is meant to scale up on a 3840x2160 screen

RetroDriven, 2022-12-23:
How do things look with the v1.3.9 release that I just published?

Kugelblitz360, 2022-12-23:
Can send you a screenshot, on my system with 150% scaling it is now broken (was not before).

Wesleyrpg, 2022-12-24:
it was a little worse with the update to 1.3.9, but i found that i can make the window bigger by dragging it open further in the bottom right hand corner of the app and now it looks great!

Wesleyrpg, 2022-12-24:
actually, the scaling is really bad in some spots.....needs more work!

RetroDriven, 2023-01-02:
Can you please upload some Screenshots where you notice issues?

Wesleyrpg, 2023-01-03:
sure
[image]

Wesleyrpg, 2023-01-03:
here....
[image]

RetroDriven, 2023-01-03:
Thanks! How do things look with v1.4.1?

Wesleyrpg, 2023-01-03:
Same problem!
[image]

Wesleyrpg, 2023-01-03:
Files and Logs are still also cut in half!

Kugelblitz360, 2023-01-12:
Just wanted to confirm that current builds (1.4.2+) look fine on my display with either 100%, 125% and 150% scaling.

RetroDriven, 2023-01-16:
Excellent thanks! I'll be releasing a revamped UI in the upcoming v1.5 soon. I'm hoping there are no weird scaling issues there.