RetroDriven/Pocket_Updater #40: Manage Cores section: save button banner hiding the bottom line of the core list

StardustX7, 2022-12-23:
Using windows 10 in 1080p resolution

RetroDriven, 2023-01-02:
Can you upload a screenshot of this please? Does it look the same when you Maximize the Window?

StardustX7, 2023-01-03:
Yes it's the same when the window is maximized, here are the screenshots
normal size [https://i.postimg.cc/N0KF9zjn/2023-01-03.png](url)
maximized [https://i.postimg.cc/5ysWmfPS/2023-01-03-1.png](url)
Thank you

StardustX7, 2023-01-03:
> Can you upload a screenshot of this please? Does it look the same when you Maximize the Window?

Yes it's the same when the window is maximized, here are the screenshots
normal size [https://postimg.cc/NywBZbT8/94a046ba](url)
maximized [https://postimg.cc/vxxKYww4/0b82014e](url)
Thank you

RetroDriven, 2023-01-03:
Thanks! How do things look with v1.4.1?

StardustX7, 2023-01-04:
> Thanks! How do things look with v1.4.1?

The manage cores menu looks perfectl now in 1.4.1, the issue is fixed thanks for your hard work.

I also noticed that Sorcer Striker and Kingdom Grandprix cores from pram0d are not listed in the manage cores menu, but they are downloaded and installed by the application, and as the others toaplan 2 cores from pram0d are all listed separately in the menu, I don't know if it's intentional or not, maybe it's just linked with the common ressources you use with the command line updater, so I just report it here in the case it can be useful, but it's absolutely not a breaking issue that needs to be fixed, just consider this as a note for consistency of the core listing.

Thanks again for all what you do.

closed 2023-01-04 by StardustX7