RetroDriven/Pocket_Updater #41: v1.3.9 scaling issues - Win10 @ 100%

esmith13, 2022-12-27:
v1.3.9 has scaling issues in Win10 @ 100% scale. Issues NEVER existed before this version

Manage Cores screen completely blocks last entry from view: [Picture](https://i.postimg.cc/xdkCTT85/image.png)

Organize Cores screen slightly clips last entry and obscures chosen drive letter completely: [Picture](https://i.postimg.cc/sx5sGMCP/image.png)

From a layman's perspective it seems to me that the bottom horizontal bar is layered overtop of the table instead of beginning after the table ends. Organize Cores doesn't clip as much as Manage Cores since it's bar shorter and thus overlaps the table less.

It is worth noting that resizing the window manually can alter how much is clipped from the last line in both mentioned windows but it's never perfect at the default size and practically impossible to resize it manually and make it mostly visible since it wants to snap the table position to mess it up again. If that makes any sense...

RetroDriven, 2023-01-03:
How do things look with v1.4.1?

esmith13, 2023-01-03:
Issues are resolved with v1.4.1.
FYI - the colored blank row at the bottom of the table in both windows exhibits the same behavior as the last line on  v1.3.9 did but since it's blank it doesn't matter. Easiest fix is often the best one.  ;)

Thank you!

closed 2023-01-04 by esmith13