RetroDriven/Pocket_Updater #43: v1.4 - jtcps1 core, Unable to find many roms in archive

Epyc7, 2022-12-30:
A bunch of jtcps1 roms are not downloading from archive:

Downloading sf2dongb.rom
Unable to find sf2dongb.rom in archive

RetroDriven, 2023-01-02:
There are a bunch of Bootleg versions that Jotego didn't include MRA files for to create the Roms. I'm looking into this in the meantime and I'll reach out to Jotego if needed.

closed 2023-01-16 by RetroDriven

cheskote, 2023-12-17:
I see this bug is closed but it seems very similar to what is happening to me.

Each time I run the updater I obtain this message at the end of the log:

```
Assets Skipped:(35)
-----------------------
captcmmb.rom, knighsb2.rom, pang3b2.rom, pang3b3.rom, sf2rules.rom, sf2rk.rom, sf2ebbl.rom, sf2stt.rom, sf2ebbl2.rom, sf2ebbl3.rom, sf2thndr.rom, sf2thdr2.rom, sf2ue.rom, sf2um.rom, sf2ee.rom, sf2ef.rom, sf2accp2.rom, sf2amf.rom, sf2dongb.rom, sf2dkot2.rom, sf2bhh.rom, sf2amf2.rom, sf2amf3.rom, sf2m2.rom, sf2m4.rom, sf2m5.rom, sf2m6.rom, sf2m7.rom, sf2m8.rom, sf2ceblp.rom, sf2rb.rom, sf2rb2.rom, sf2koryu.rom, sf2yyc.rom, varthb2.rom, 
```

Is anyone else experiencing this behavior?