RetroDriven/Pocket_Updater #46: Download doesn't start

Grabou, 2023-01-07:
Version : 1.4.2

Will try to reboot my computer later but...

🤔🤔🤔

[image]

brownby, 2023-01-08:
Had the same problem, downloads work upon downgrading to 1.3.9

IcedEagle, 2023-01-09:
Just downloaded this for the first time and same issue. Only downloads the firmware, and it never downloads any core or BIOS. Tried un-checking all cores, retrying and then checking all cores to download and retrying again. Still no dice.

RetroDriven, 2023-01-10:
Did you have issues with other releases? I can't reproduce this issue on my end. I'm wondering if your firewall/router may be blocking something. Do you have any type of content filtering setup?

brownby, 2023-01-10:
This is the first time I've used the updater unfortunately (my Pocket is arriving this week so I was just getting all the cores ready to go on an SD card in preparation), so I don't have any past experience to compare to. 

I don't have anything other than Windows Defender running on the machine I tried this on, my router is from my ISP so it's possible there's something being blocked there I suppose. 

A couple of more pieces of info that may or may not be helpful:

- I tried running it as an admin in case it was some kind of permissions issue, no dice
- As @IcedEagle mentioned, it _does_ download the firmware, then just hangs on downloading any cores
- Before downgrading to 1.3.9 I also tried downloading Asset Image Packs and that worked fine as well
- I tried adding my GitHub personal access token and saw the same behavior

Grabou, 2023-01-10:
This did not happen with older versions of the software, but when I tried again yesterday with an older version, same behavior. So it is possible that the problem is elsewhere.

I do have a router that blocks things + a network-wide ad blocker (AdGuard Home), I'll check that it's not them that block.

IcedEagle, 2023-01-11:
@RetroDriven Found the issue. It's because the app didn't create the folder called "Cores", but it needs that to search for the cores to update.

In Updater.cs : GetLocalCores, the line for         string[] directories = Directory.GetDirectories(coresDirectory,"*", SearchOption.TopDirectoryOnly);

This is throwing a DirectoryNotFound exception saying path not found (because the coresDirectory itself doesn't exist), but because of async shenanigans, the exception is getting lost and this deadlocks.

RetroDriven, 2023-01-11:
@IcedEagle Good find! There was a bug fix for this with @mattpannella's updater that I didn't see before I pushed v1.4.2 but this should be all fixed up with v1.4.3 that I just pushed out.

Grabou, 2023-01-11:
It's working with 1.4.3, thanks! :)

closed 2023-01-11 by Grabou