RetroDriven/Pocket_Updater #47: Missing some Amiga files - Log does not contain missing files

Kugelblitz360, 2023-01-12:
I see that some Amigafiles (fasthd/slowhd) are not found, also there seems to be a missing palette file for the NES core. Unfortunately the log files do not track which files were missing during an update run.

RetroDriven, 2023-01-14:
Those missing Amiga files are not required and really shouldn't be trying to download. There will be a Blacklist of known files that shouldn't be trying to download in the next release so this can be ignore for now. I will also be adding a a section to the Log file for items that are missing but couldn't download as well.

What NES Pallette are you talking about?

Kugelblitz360, 2023-01-14:
I would like to give you more details, but stuff scrolls by and you can not scroll back and it is not written to the LOG but I think it is a .pal file for Spiritualized NES that is not found.