RetroDriven/Pocket_Updater #49: Firmware Update detection issue

esmith13, 2023-01-14:
So the way I personally use the Pocket Updater is to keep it in my PC Hard Drive and update to "Current Directory" first (as a backup & test) and then I run the update again against my SD card (Removable Storage).

When I updated just now (current directory on HDD) it found the new Pocket firmware update and downloaded it to my HDD. I then put my SD card in my PC and ran the updater again on "Removable Storage" and it says there were no updates found. I quit and relaunched the app and it still didn't find any updates. This time I paid close attention and it said my firmware was already up to date.

Somehow if the firmware update is already in the "current directory" (my PC hdd) it decides my SD card already has it as well, even if it doesn't.

I did not experience this in a previous version of the app and I have no idea if that is because the older versions didn't have the issue or just the timing of firmware releases prevented it from presenting itself.

RetroDriven, 2023-01-14:
Hi - Can you see if v1.4.4 resolves this issue when possible? I believe it should.

esmith13, 2023-01-16:
> Hi - Can you see if v1.4.4 resolves this issue when possible? I believe it should.
I would say it's fixed in v1.4.4 if the following steps are considered a valid test:
- Removed current FW file from SD card
- update "current directory" on local HDD (newest FW is now present there)
- update SD card

In that scenario it did find and add the current FW to the SD card instead of saying it was already downloaded.
If the above is not a valid test, I will be sure to test in the same order when the next Pocket FW releases.

Thank you!

closed 2023-01-16 by RetroDriven