RetroDriven/Pocket_Updater #54: Selecting Removable Storage -> Update results in drive not found

DeerSteak, 2023-02-23:
If you select Removable Storage, it selects the first removable drive on the system. If that happens to be your SD card and you press Update, you'll get a "The Drive Letter Was Not Found!" message. Press update again and it updates as expected. 

[image]
[image]

RetroDriven, 2023-02-23:
I'll do some testing and fix it. The GitHib Token box and Save buttons look off for you as well. Those should be directly below the Use GitHub Token setting.

RetroDriven, 2023-02-23:
I found/fixed the issue and I'll put that fix in place for the next release.

DeerSteak, 2023-02-23:
That's awesome, thanks. BTW don't worry about the arrangement too much, I'm using 175% scaling on a 28" 4K monitor, so I'm used to stuff not looking exactly right.

RetroDriven, 2023-02-25:
You're welcome.

closed 2023-02-25 by RetroDriven