RetroDriven/Pocket_Updater #55: Check boxes for individual cores not appearing under “Manage Cores”

Gorgonner, 2023-02-24:
Check boxes for individual cores not appearing under “Manage Cores”. Unable to select cores.

only check box is the one for all in the blue header

occurring in 1.5.0 and 1.5.1

surface pro 7 
Win 10
![9E607F59-0F17-406D-9B8A-38A7C3C1EABD](https://user-image
[image]
s.githubusercontent.com/126216033/221075803-a7cf4afa-90bd-4d43-9a34-377997c9e662.jpeg)

i5 16gb ram

Gorgonner, 2023-02-24:
[image]

RetroDriven, 2023-02-24:
Can you upload a screenshot of what you're seeing?

Gorgonner, 2023-02-24:
[image]

RetroDriven, 2023-02-25:
Definitely an odd one. Can you see if the issue remains in v1.5.2?

dgarra, 2023-02-25:
Can confirm still problematic in 1.5.2.

RetroDriven, 2023-02-25:
It definitely sounds like a scaling thing. Can you both share what Resolution and DPI/Scaling you have set?

Gorgonner, 2023-02-25:
1.5.2

Using the surface pro 7 default/“recommended” resolution and scaling:
2736 x 1824
Scale: 200%

Still present on 1.52 with the above resolution and scaling.

The check boxes do appear when I set scaling to 100% (but everything on my screen is then very tiny)

dgarra, 2023-02-26:
I'm running mine out of a virtual machine, so the scaling is dynamic via the size of the VMware window. If i bring the window to a 1920x1080 display, where the scaling is 100%, the checkboxes show up. Anything larger, like my main 5k display, where the scaling goes above 200%, the gui shows the behavior.

RetroDriven, 2023-02-27:
Hi - I'm hoping this has all been resolved in v1.5.4

Please check and let me know when possible please.

Gorgonner, 2023-02-27:
Fix verified on 1.5.4

Thanks for the fix, now I can continue to use my favorite updater 🙂

RetroDriven, 2023-02-27:
Yay! ☺️

closed 2023-02-27 by RetroDriven

dgarra, 2023-02-27:
Confirmed resolved in 1.5.4. Thanks and great work on this.