RetroDriven/Pocket_Updater #57: Updater becomes completely unresponsive

Missingno255, 2023-02-25:
After attempting to update my Pocket, using the latest version of the updater, the program completely crashes and becomes completely unresponsive. You cannot kill the process at all. Using Task Manager, Task kill command in Command Prompt, Process explorer to kill process/end task is useless. Process explorer says access is denied, even when it's using admin privileges. Windows itself can't even terminate it, when the "waiting for program to respond" box comes up. After it "reports the problem to Microsoft", it still doesn't close out and sits there completely crashed. When Windows attempts to restart, it prevents the OS from restarting properly, because it cannot terminate the program at all, and you essentially have to hard reboot/use reset button.

RetroDriven, 2023-02-25:
Can you do a screen record of the behavior and share it? Do you have any other AV software installed or just Windows Defender? I have never seen this happen with all my testing so I'm not sure what's going on here for you.

SpiralBrad, 2023-02-26:
> After attempting to update my Pocket, using the latest version of the updater, the program completely crashes and becomes completely unresponsive. You cannot kill the process at all. *snip*

This won't fix the issue at hand, but the next time it goes unresponsive...

Try: `taskkill.exe /f /im PocketUpdater.exe` from the Win+R run box or an (elevated, if you wish) command prompt.

That can often be a magic bullet to kill a badly behaving app which other things can't seem to kill.

YLFAndy, 2023-03-24:
> Can you do a screen record of the behavior and share it? Do you have any other AV software installed or just Windows Defender? I have never seen this happen with all my testing so I'm not sure what's going on here for you.

Happening for me as well. It looks like maybe there's a memory leak. I've never had it happen on my wife's pocket, but she has a limited set of roms. I have all the things (it's a hoarding disorder, I'll never play most of it). 

Killing the process and restarting eventually works, but I have to do it 2 or 3 times usually. More as the library of arcade games grow. I'll try to grab a video next time I update (I already went through that process today for the JT Friday stuff and had to restart the process 3 times).

Missingno255, 2023-03-25:
> > Can you do a screen record of the behavior and share it? Do you have any other AV software installed or just Windows Defender? I have never seen this happen with all my testing so I'm not sure what's going on here for you.
> 
> Happening for me as well. It looks like maybe there's a memory leak. I've never had it happen on my wife's pocket, but she has a limited set of roms. I have all the things (it's a hoarding disorder, I'll never play most of it).
> 
> Killing the process and restarting eventually works, but I have to do it 2 or 3 times usually. More as the library of arcade games grow. I'll try to grab a video next time I update (I already went through that process today for the JT Friday stuff and had to restart the process 3 times).
> 
> EDIT: after doing some more tests, it appears to ONLY happen when my microSD is connected using an SD adapter. Using a reader with a microSD slot doesn't hang. Tried multiple adapters from SanDisk, Kingston, Lexmark, and Crucial as well as multiple readers (Kingston and ProGrade).

Well, I guess I'm screwed, I don't have a reader with a Micro SD card slot, and this laptop I have, only has a standard SD slot.

@SpiralBrad even doing that taskkill.exe /f /im PocketUpdater.exe doesn't work. Nothing does. Only way is to restart, and it prevents Windows 10 from restarting properly.

SpiralBrad, 2023-03-25:
@Missingno255 Oh oops, I see now that you tried that in the initial post.  Not sure how I glossed over that, hah.

And I see you looked at Process Explorer, but maybe try Process _Monitor_ and watch absolutely everything it does... and take note of the last few things it tries to touch/do before freezing?

YLFAndy, 2023-03-25:
> Well, this is a problem, I don't have a reader with a Micro SD card slot, and this laptop I have, only has a standard SD slot.
> 
> @SpiralBrad even doing that command (which was one of the first things I tried, doesn't work. Nothing does. Only way is to restart, and it prevents Windows 10 from restarting properly.

This is going to probably be a 'do it before bedtime' thing, but have you tried running the updater off of the Pocket directly with USB connection? I did that before CD and arcade cores hit and never had a problem. But it's, obviously, not ideal. If it doesn't freeze, though, it might support a race condition theory. I tested the throughput on my readers and when it's in the microSD reader it's about half the speed as when it's in an SD adapter/reader.

Missingno255, 2023-03-25:
> This is going to probably be a 'do it before bedtime' thing, but have you tried running the updater off of the Pocket directly with USB connection? I did that before CD and arcade cores hit and [...]

I tried that with a USB-A to C cable that my smart phone came with, and it only charges, no data connection. I don't have a converter to use Analogue's horrible USB-C only cable.

RetroDriven, 2023-04-04:
Some major changes were made in v1.5.6 with Downloads/Speeds. Is this issue still happening?

YLFAndy, 2023-04-04:
>Some major changes were made in v1.5.6 with Downloads/Speeds. Is this issue still happening?
Still happening for me if I use an SD card adapter, but now it's actually crashing rather than just hanging. 

[image]

RetroDriven, 2023-04-04:
What happens if you disable CRC checking? Are you clicking the UI while it Updates or just letting it sit?

YLFAndy, 2023-04-04:
> What happens if you disable CRC checking? Are you clicking the UI while it Updates or just letting it sit?

Clicking. Let me disable CRC And try again.

RetroDriven, 2023-04-04:
It's not really going to like clicking around the UI while it's downloading as it is now. Try letting it sit and update instead to see what happens.

YLFAndy, 2023-04-04:
> It's not really going to like clicking around the UI while it's downloading as it is now. Try letting it sit and update instead to see what happens.

Oh, no...I clicked it AFTER it hung.

RetroDriven, 2023-04-04:
Where is it hanging up at? Is it the same place every time?

I'm trying to reproduce this but so far I haven't been able to. What kind of specs do you have as far as the laptop/desktop?

Are you running the Updater from the machine itself or is it saved/running from the MicroSD?

YLFAndy, 2023-04-04:
> Where is it hanging up at? Is it the same place every time?
>

It differs. It normally happens in the JT cores, whether there's anything to download or not, but on the most recent build it was happening earlier. 
 
> I'm trying to reproduce this but so far I haven't been able to. What kind of specs do you have as far as the laptop/desktop?
> 
Intel 10980XE, 256GB DDR4, Win11 x64, system drive 2TB Samsung 980Pro NVMe Gen4 M.2
But it also happens on my laptop, which is a 12th gen Intel with 64GB RAM. 
My readers are Kingston FCR-N84, ProGrade USB 3.2 CFExpress/SD UHS-II, Intel integrated, Corsair.

> Are you running the Updater from the machine itself or is it saved/running from the MicroSD?
From the MicroSD. 

CRC disabled DID improve it massively in speed and it didn't crash. Cloning the card and running it from SSD works fine. I have not tried just keeping the updater on my machine and running it outside of current directory. I'll give that a go as well.

YLFAndy, 2023-04-04:
Running it from my PC against removable storage versus directly from the MicroSD seems to have remedied it. I'll do that from now on. Unfortunately nothing useful shows up in eventviewer, so I don't think I can add much more.

closed 2023-11-14 by RetroDriven