RetroDriven/Pocket_Updater #58: Scaling issues

SWEiNSANE, 2023-02-26:
I'm running Windows 11 and since I started using this excellent app, scaling issues have been there. My laptops resolution is 3840 x 2400 pixels and therefore I'm running 225% zoom (scaling).

The issues makes:

- the checkboxes when managing the cores to disappear
- buttons for minimize, maximise and close are really small
- texts are looking wierd and not shown correctly
- can't change the size of the window by dragging the corners
- can't choose drive letter for organizing cores

**Crippled user interface**
[image]

**No checkboxes when managing cores**
[image]

**Cannot organize cores**
[image]

RetroDriven, 2023-02-27:
Hi - I'm hoping this has all been resolved in v1.5.4

Please check and let me know when possible please.

SWEiNSANE, 2023-02-27:
Hi,

Sounds great and thanks for an excellent app

SWEiNSANE, 2023-02-27:
I've now tried v1.5.4 and now everything works as expected - thanks and keep up the good work!

**Update Pocket**
[image]

**Manage Cores**
[image]

RetroDriven, 2023-02-27:
Good deal! You're welcome.

closed 2023-02-27 by RetroDriven