RetroDriven/Pocket_Updater #60: Asset Image Pack downloader looks broken

uberyoji, 2023-02-28:
I am using version 1.5.4.

Here is the error log:

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.IO.DirectoryNotFoundException: Could not find a part of the path '\\?\C:\Users\jsavage\AppData\Local\Microsoft\VisualStudio\17.0_1fa20820\WinFormsDesigner\421upsnr.pvw\imagepack.zip'.
   at Microsoft.Win32.SafeHandles.SafeFileHandle.CreateFile(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.Open(String fullPath, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.File.OpenHandle(String path, FileMode mode, FileAccess access, FileShare share, FileOptions options, Int64 preallocationSize)
   at System.IO.File.WriteAllBytes(String path, Byte[] bytes)
   at pannella.analoguepocket.HttpHelper.DownloadFileAsync(String uri, String outputPath)
   at pannella.analoguepocket.ImagePack.fetchImagePack(String path)
   at pannella.analoguepocket.ImagePack.Install(String path)
   at Pocket_Updater.Controls.Image_Packs.Image_Packs.dataGridView1_CellContentClick(Object sender, DataGridViewCellEventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)

************** Loaded Assemblies **************
System.Private.CoreLib
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
Pocket Updater
    Assembly Version: 1.5.4.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Runtime
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Windows.Forms
    Assembly Version: 6.0.2.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.ComponentModel.Primitives
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Windows.Forms.Primitives
    Assembly Version: 6.0.2.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Runtime.InteropServices
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Drawing.Primitives
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Collections.Specialized
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Threading
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
[... 252 lines cut ...]
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Console
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Diagnostics.StackTrace
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Reflection.Metadata
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------
System.Collections.Immutable
    Assembly Version: 6.0.0.0
    Win32 Version: n/a
    CodeBase: 
----------------------------------------

************** JIT Debugging **************

RetroDriven, 2023-03-01:
This should be resolved in v1.5.5 that I just released. Can you please confirm?

uberyoji, 2023-03-01:
Solved! Thx.

closed 2023-03-01 by uberyoji