RetroDriven/Pocket_Updater #63: BUG: Timeout while actively downloading files.

dot-bob, 2023-03-26:
I have been getting the error "The request was canceled due to the configured HttpClient.Timeout of 100" while it is actively downloading data.

I can verify the download is active as the task manager shows Pocket Updater is using 1-4Mbps of network traffic.  It requires me to run the updater over and over again until I get a data rate that can complete the download before timing out.

TheMrChao, 2023-03-27:
+1

...seems to happen with many Jotego cores while scanning external media.

Unable to read E:\Assets\jtvigil\jotego.jtvigil\Vigilante (US, Rev G).json
The request was canceled due to the configured HttpClient.Timeout of 100 seconds elapsing.

uberyoji, 2023-03-30:
I think this is because the roms for cps2 are much bigger and since they seem to come from archive.org (which is slow sometimes), they take more than 100s to download and timeout...

RetroDriven, 2023-04-04:
Some major changes were made in v1.5.6 with Downloads/Speeds. Is this issue still happening?

uberyoji, 2023-04-04:
Looks fixed on my end. Blazing fast now.
Thanks!

RetroDriven, 2023-04-04:
Excellent!

closed 2023-04-25 by RetroDriven