RetroDriven/Pocket_Updater #66: Update button allows for multiple clicks...

SpiralBrad, 2023-05-04:
The main 'Update' button allows a user to double/triple/etc -click the button, seemingly starting multiple update processes, as the status text on the right is doubled/tripled.

[image]

I wonder if this is related to #57; perhaps those users are double-clicking and not realizing or stating such?

RetroDriven, 2023-09-08:
Is this still happening with v1.5.7?

SpiralBrad, 2023-09-08:
I can confirm that this no longer happens as of v1.5.7 on the same Windows 10 system where it was previously encountered.

SpiralBrad, 2023-09-08:
OOPS, spoke too soon.  With a triple click, it seems like it behaved differently, but I was seeing DOUBLE log entries, but then also a triple "No updates found" dialog at the end (had to click OK 3 times).

I'm unsure why I was seeing double log entries as opposed to triple, unless the 'timing' was off and I just didn't notice.

Ending of my log looks as follows, after what I assume was registered as a triple click of "Update" to the Current Directory (an SD card):
```
-------------
-------------
Checking Core: Spiritualized.Megaduck
Checking Core: Spiritualized.Megaduck
1.0.2 is the most recent release, checking local core...
1.0.2 is the most recent release, checking local core...
local core found: 1.0.2
local core found: 1.0.2
Looking for Assets
Looking for Assets
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.NES
Checking Core: Spiritualized.NES
1.0.2 is the most recent release, checking local core...
1.0.2 is the most recent release, checking local core...
local core found: 1.0.2
local core found: 1.0.2
Looking for Assets
Looking for Assets
Already installed: palettes/ntsc.pal
Already installed: palettes/ntsc.pal
Already installed: fds.bios
Already installed: fds.bios
Already installed: C2C03.PAL
Already installed: C2C03.PAL
Already installed: C2C03B.PAL
Already installed: C2C03B.PAL
Already installed: P2C04_1.PAL
Already installed: P2C04_1.PAL
Already installed: P2C04_2.PAL
Already installed: P2C04_2.PAL
Already installed: P2C04_3.PAL
Already installed: P2C04_3.PAL
Already installed: P2C04_4.PAL
Already installed: P2C04_4.PAL
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.Odyssey2
Checking Core: Spiritualized.Odyssey2
1.0.2 is the most recent release, checking local core...
1.0.2 is the most recent release, checking local core...
local core found: 1.0.2
local core found: 1.0.2
Looking for Assets
Looking for Assets
Already installed: o2bios.bin
Already installed: o2bios.bin
Already installed: SP128_03.BIN
Already installed: SP128_03.BIN
Already installed: SP128_04.BIN
Already installed: SP128_04.BIN
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.SG-1000
Checking Core: Spiritualized.SG-1000
1.2.0 is the most recent release, checking local core...
1.2.0 is the most recent release, checking local core...
local core found: 1.2.0
local core found: 1.2.0
Looking for Assets
Looking for Assets
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.SMS
Checking Core: Spiritualized.SMS
1.2.0 is the most recent release, checking local core...
1.2.0 is the most recent release, checking local core...
local core found: 1.2.0
local core found: 1.2.0
Looking for Assets
Looking for Assets
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.Studio2
Checking Core: Spiritualized.Studio2
1.0.2 is the most recent release, checking local core...
1.0.2 is the most recent release, checking local core...
local core found: 1.0.2
local core found: 1.0.2
Looking for Assets
Looking for Assets
Already installed: rca2bios.bin
Already installed: rca2bios.bin
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.SuperGB
Checking Core: Spiritualized.SuperGB
1.0.3 is the most recent release, checking local core...
1.0.3 is the most recent release, checking local core...
local core found: 1.0.3
local core found: 1.0.3
Looking for Assets
Looking for Assets
Already installed: sgb_snes.smc
Already installed: sgb_snes.smc
Already installed: sgb_boot.bin
Already installed: sgb_boot.bin
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.Supervision
Checking Core: Spiritualized.Supervision
1.0.0 is the most recent release, checking local core...
1.0.0 is the most recent release, checking local core...
local core found: 1.0.0
local core found: 1.0.0
Looking for Assets
Looking for Assets
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
Checking Core: Spiritualized.VideoBrain
Checking Core: Spiritualized.VideoBrain
1.0.0 is the most recent release, checking local core...
1.0.0 is the most recent release, checking local core...
local core found: 1.0.0
local core found: 1.0.0
Looking for Assets
Looking for Assets
Already installed: uvres1.bin
Already installed: uvres1.bin
Already installed: uvres2.bin
Already installed: uvres2.bin
Up to date. Skipping core
Up to date. Skipping core
-------------
-------------
```

esmith13, 2023-11-02:
I have the same issue. updates were slow to start the other day and I clicked the button a couple more times. Every command was repeated a total of 3 times.

DiskJunky, 2023-11-07:
The PR [here ](https://github.com/RetroDriven/Pocket_Updater/pull/77) should hopefully address it if it gets merged.

RetroDriven, 2023-11-09:
@DiskJunky Thanks for doing that! Please check and see if the issue is resolved with v1.5.8

esmith13, 2023-11-09:
v1.5.8 has fixed this issue for me. 
I could reproduce the problem consistently on v1.5.7 and cannot force it to happen at all in v1.5.8.

DiskJunky, 2023-11-09:
@RetroDriven, confirmed addressed in v1.5.8

closed 2023-11-14 by RetroDriven