RetroDriven/Pocket_Updater #67: Updater ignores selected cores

tylerfox3894, 2023-05-05:
I specifically select and un-select a list of cores i want, save the option, and then update. The program simply ignores my choices and downloads every single core

DakotaMcCutch, 2023-05-11:
> I specifically select and un-select a list of cores i want, save the option, and then update. The program simply ignores my choices and downloads every single core

Just don't hit the save button on the Update Pocket Screen, that wipes out the configuration. The other ones seem fine. I fixed it on my system. Meant to do a PR before I left. To help the devs out.

theory18, 2023-11-24:
> @tylerfox3894 I did a PR for a temp fix; I found a bug that causes the core settings to be wiped out.
> 
> It only happens the first time the application is used and a new settings file is created. Something to do with the Setting Managers reference not being set correctly; I produced the same bug once or twice when it loaded a pre-existing settings file. But it was challenging to do.

Hi, i'm trying to use your app (thanks for creating this), but i stumble across the same issue.
I select the cores i want, press save on the "Manage Cores" tab then when i press update, it still installs all the cores.
What is the general steps to take, because i'm confused.
Thanks in advance

holloway87, 2023-11-28:
I have the same problem, it completely ignores my selection. After an update new cores were available, i deselected them and it still downloaded them.
I then deleted the configuration file and started fresh, selected all the cores i wanted, the app installed all available cores instead. How can we fix this?

holloway87, 2023-11-28:
Okay, i found steps to get it running with the selection i want:
1. Open the app and select all cores you want, then hit Save in the Manage Cores view.
2. Close the app and open it again, then press Update in the Update Pocket view.

RetroDriven, 2024-05-12:
Is this issue still happening with v1.6.1?

closed 2024-05-12 by RetroDriven