RetroDriven/Pocket_Updater #70: Specified cast is not valid.

ah394496, 2023-07-17:
Hello,
When I try to change which cores I would like to keep/update, the app does not allow me to make changes to the list. If I toggle on all of them, the app works as it should and checks/updates them all. But if I go in and uncheck some, I am not permitted to save my changes and am told 'Specified cast is not valid'. Am I doing something wrong? I am using the latest version on a PC running Windows 10.
My thanks,
Andy

frozenwizard, 2023-08-18:
I had the same issue.  After installing things to the pocket and restarting the app, I did not get that error afterwards.

ah394496, 2023-08-19:
Thanks for the suggestion, on my end I finally got the message to go away. In the list of cores I was trying to check or uncheck many at once. The app didn't like that. But if I did it one at a time, saving after each time, it seemed to work.

RetroDriven, 2023-09-08:
Is this still happening with v1.5.7?

ah394496, 2023-09-08:
Hello,
It's not happening if I select cores one at a time. So I have ten that I no longer wish to have, I have to uncheck one, save, uncheck one, save, etc. If I try to do this in a batch manner, that's when I get the error.
Andy

jzrola, 2023-09-23:
I have the same problem. My workaround was to manually change the skip element from true to false or vice versa.

ah394496, 2023-09-24:
Interesting. I appreciate you getting back to me on this. A couple nights ago I updated my Pocket with Updater. Jotego's Sega16 core was added. Have you tried that one yet? It loads, but any time I select a game within the emulator, I'm met with a Patreon nag screen I can't get rid of. I was curious if you've seen that as well. I don't know how to get around it. I wrote to him via Twitter (X), but no response.
Andy

ah394496, 2023-10-01:
Quick update, the new version of the Sega16 core has the Patreon nag screen removed. The new Teenage Mutant Ninja Turtles core, however, did have a nag screen and I can't figure out how to turn it off. Thanks for all you do, it's appreciated

ivodopiviz, 2023-10-22:
This error is because the DataGrid cells seem to have a string value instead of boolean in some cases, but I haven't been able to reproduce why. A faster repro case is to click quickly the same checkbox over and over until you crash the app. At least it crashes to the debugger instead of giving you the cryptic message box.

ah394496, 2023-10-22:
I appreciate you sticking with this.Some of the recent joetgo cores added in the app have nag screens that won't go away... do you know how to turn them off? Looks like Patreon credits or something

RetroDriven, 2024-05-12:
This should be resolved with v1.6.1

closed 2024-05-12 by RetroDriven