RetroDriven/Pocket_Updater #71: Feature Request: Category on Manage Cores page

s-taylor, 2023-08-30:
Firstly, thankyou for your awesome work on this!

I have a minor feature request, it would be awesome if we could see the "Category" on the Manage Cores page. When I'm choosing which cores to enable/disable I typically want to enable all the Arcade cores and just disable Handhelds/Consoles I'm not interested or don't have the roms for. But it's a little hard to know which is which without the Category.

z3razerviper, 2024-07-27:
I agree that would make this app even better.

RetroDriven, 2026-09-18:
This will be added to a v2 of the App that I've been working on. I'll be releasing this over the weekend 

[image]

RetroDriven, 2026-09-18:
V2 has been released. Enjoy 😊

closed 2026-09-18 by RetroDriven