RetroDriven/Pocket_Updater #73: Delete Skipped Cores does not work

Squish-the-Android, 2023-09-10:
This may be related to the other issue posted where a user is having trouble with the manage cores screen.  Delete Skipped Cores simply doesn't delete unselected cores.  It just leaves them.  No errors present when running the update.

jay8ee, 2023-10-18:
I encountered this too.

I did a little investigating. If you select the cores you want then save, you can see in the preferences .json that the core settings are correct. If you then go back to the main settings page and make any changes and save, the core settings are removed from the preferences.

ivodopiviz, 2023-10-22:
I was so upset at this that I forked the repo and started looking into it, but I'll pass on this one. If I was a betting man, I'd look into the async methods that aren't awaited anywhere. Problem is, running this in a debugger will mask those async shenanigans, so good luck.

nortakales, 2023-10-22:
Unfortunately this utility is basically useless with such a huge issue. I stopped using it and moved to https://github.com/mattpannella/pocket-updater-utility instead.

RetroDriven, 2023-11-09:
Please check and see if the issue is resolved with v1.5.8

Squish-the-Android, 2023-11-11:
> Please check and see if the issue is resolved with v1.5.8

I'm getting the same issue.  It's just resetting the pocket_updater_settings.json core settings blank.  It looks like it resets it to blank when you hit the save button on the main page OR you hit the update button.

Squish-the-Android, 2023-11-11:
Sorry about closing this, its still open.

phyber, 2023-11-28:
I'm currently watching the updater download a bunch of jotego cores even though every single one is unchecked.

It seems like it does a scan of the Assets directory, and if it finds anything, it downloads it. Completly ignoring the `skip` setting. Obviously, it should also have deleted these directories before this point too.

On this topic, a "skip new cores by default" should probably be added, that's probably a different feature request though.

Squish-the-Android, 2023-11-28:
It doesn't just ignore the setting.  It wipes out all the skip settings.
You can look at the text file before and after hitting the update button.

phyber, 2023-11-28:
My skip settings are still in the file, for every single core I want skipped (on Pocket Updater v1.5.9), they just seem to be ignored.

ericmorand, 2023-12-10:
Same issue here, deselecting some cores - actually selecting only the SNES one - and the app still download every core available.

RetroDriven, 2024-05-12:
Is this issue still happening with v1.6.1?

closed 2024-05-12 by RetroDriven