RetroDriven/Pocket_Updater #74: Fix Jotego Core Names not working.

SpookySquidGames, 2023-09-24:
I've been trying to get the "Fix Jotego Core Names" option to work without any success.

I've tried it with and without Preserve Platforms turned on. I've tried removing the contents of the Platforms directory and all my cores manually and reinstalling them with Pocket Updater. The Jotego Core names just aren't being changed. 

Is there some trick to getting this to work?

SpookySquidGames, 2023-11-09:
No luck. They still have the standard names after running an update. Anything special I might need to do to make this work?

RetroDriven, 2023-11-20:
I believe this has been fixed now in v1.5.9. Give this a try when possible please.

SpookySquidGames, 2023-11-25:
Still no luck. I'd actually used Pocket Sync to rename these recently (since my previous post). But to test I went and deleted and reset the jtninja core to its default name and category. Ran pocket updater a few times and no luck, the name just wouldn't update. I wonder what it is about my setup is causing it not to work for me?  I assume I'm not the only person running into this problem? Let me know if there is any helpful info, files I can provide etc.

RetroDriven, 2024-05-12:
Is this issue still happening with v1.6.1?

closed 2024-05-12 by RetroDriven

SpookySquidGames, 2024-05-13:
Sorry was away all weekend so only had a chance to test today. Still no luck. Cores like jt1943, jtcommnd, and all the others I haven't renamed with other tools, are still not being renamed with v1.6.1. 

Does the renaming show up in the update log? Maybe including that info might help track down the issue.