RetroDriven/Pocket_Updater #75: Unable to open URLs in "Asset Image Packs" tab without selecting a download location

demize, 2023-10-10:
When first opening the Asset Image Packs tab, no download location is selected. Trying to open any of the Github URLs for the asset packs with no download location selected results in an unhandled (I believe null pointer; not sure) exception.

UScr00ge, 2023-11-17:
I'm getting the same issue when hitting the download button for any of the asset image packs.

Error details below:
```
************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at Pocket_Updater.Controls.Image_Packs.Image_Packs.Save_Preferences_Json()
   at Pocket_Updater.Controls.Image_Packs.Image_Packs.dataGridView1_CellContentClick(Object sender, DataGridViewCellEventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)

```

RetroDriven, 2024-05-12:
Is this issue still happening with v1.6.1?

closed 2024-05-12 by RetroDriven

JackWitherell, 2024-09-15:
Still no good: 

[image]