RetroDriven/Pocket_Updater #76: Feature Request: Skip Downloading "*.sav" files

esmith13, 2023-11-02:
I would like to request a toggle to skip attempting to download and *.sav files as part of an update.
I am one of the likely few people who keep the "Skip Alternative Arcade Files" option OFF so that I get these alternates, but that also means a RIDICULOUS amount of wasted time waiting for 98% of the roms to have a failed attempt at downloading a matching ".sav" file. Not only does this at least triple the time an already long update takes, but even if I'm fully up to date and run the updater again (let's say one core was updated) it still tries to download ALL the ".sav" files again for EVERYTHING since it never found them on a previous run.

Thank you for your awesome work and your consideration of this request.

[image]

esmith13, 2023-11-02:
today specifically, I have been "updating" for 2.5hrs so far and it is still not complete yet... It is steadily progressing, just taking many, many seconds per "problem" item such as sav files or roms that aren't available at updater.retrodriven.com

mattpannella, 2023-11-08:
I fixed this issue a few months ago so once he makes a new build with my changes from upstream it should be resolved

RetroDriven, 2023-11-09:
Please check and see if the issue is resolved with v1.5.8

esmith13, 2023-11-09:
absolutely fixed for me! Thank you both so much!

My last update with "Skip Alternative Files" turned off took over 2.5 hours. Today, using v1.5.8 it took under 3 minutes!

closed 2023-11-14 by RetroDriven