RetroDriven/Pocket_Updater #80: Unchecked Cores Being Downloaded

HaydenCupcake, 2023-12-21:
Hi,

Pocket Updater v1.6.0 seems to be downloading cores that are not checked. Please see the attached log and screenshot. For example, PokemonMini is unchecked but Pocket Updater still downloads it. I don't recall the previous version having this issue.
[pocket_update_log_122123.txt](https://github.com/RetroDriven/Pocket_Updater/files/13746589/pocket_update_log_122123.txt)
[image]
[image]
[image]
[image]

mario-64, 2023-12-24:
I can confirm this issue.

wadefree85, 2024-01-03:
also encountered this issue.  I only use a few cores and would like to only select the ones that I use

preppietechie, 2024-01-06:
Confirmed... I now have a ton of cores I don't use installed. Is there a way to uninstall them?

uberyoji, 2024-01-31:
Same problem. @preppietechie I use Pocket Sync to remove those individually.

Arq24, 2024-02-05:
Same 
App is basically usless
And looks like is dead
Thx anyway

gabguy, 2024-02-11:
Pressing the "Update" button resets the JSON file where the core preferences are saved.
Please fix this

mario-64, 2024-02-11:
Pretty sure this app has been abandoned.

preppietechie, 2024-02-13:
> Pretty sure this app has been abandoned.

Eh, people have lives. Things happen. The code is open, so any one of us could fork it and continue the work. I’m using Pocket Sync for now. I don’t like it as much as pocket updater, but it gets the job done.

https://github.com/neil-morrison44/pocket-sync

RetroDriven, 2024-05-12:
Is this issue still happening with v1.6.1?

closed 2024-05-12 by RetroDriven