RetroDriven/Pocket_Updater #81: Updater stops responding.

k5jat, 2023-12-24:
The Updater, when I went to run it today, stopped responding. It gets just past the section in the screenshot and just stops responding.

[image]

dreimer1986, 2023-12-26:
Did you happen to click around in other apps and windows? This happened quite often to me as soon the the updater loses focus as it seems...

k5jat, 2023-12-26:
> Did you happen to click around in other apps and windows? This happened quite often to me as soon the the updater loses focus as it seems...

It just stops doing anything. I've left it running for 10 minutes without touching it and nothing. Task Manager shows zero activity and clicking in the app causes Windows to figure out it's not responding, so you're right about that.

RetroDriven, 2024-05-12:
Is this issue still happening with v1.6.1?

closed 2024-05-12 by RetroDriven