RetroDriven/Pocket_Updater #84: Openfpga gba core ，the crt display mode is not supported

Bladerunner2024, 2024-02-13:
Hello,my analogue pocket os version is v2.1
But i found there was only normal display mode supported when i was playing games on openfpga gba core platform.
I can’t fix it 
Thank u very much if u could do me a favor

Dawbo66, 2024-02-14:
I don't use this updater as it crashes a lot, but i know that once you've updated you might have to install the filters again, i use pupdate and there's an option for it, that's probably why you've lost the filters.

Bladerunner2024, 2024-02-15:
> I don't use this updater as it crashes a lot, but i know that once you've updated you might have to install the filters again, i use pupdate and there's an option for it, that's probably why you've lost the filters.

Thanks a lot. I will have a try👍

closed 2024-05-12 by RetroDriven