RetroDriven/Pocket_Updater #89: Default setting installs a ton of unwanted cores, no cancel available.

peeweek, 2024-06-16:
I had a quite small system made of few cores (NES, GB, GBC, GBA, GEN, SFC)  and after clicking update, it started downloading a ton of jotego cores, with their respective roms. I don't understand why these cores are enabled by default, should the software assume that it should not install cores, unless opt in?

Moreover, I can't cancel the update, so I had to wait a long time for it to install all these unwanted cores. so i could remove them afterwards.

kewalker, 2026-09-16:
Implemented in https://github.com/RetroDriven/Pocket_Updater/pull/98

Missing cores previously had no saved disabled/unselected state, so they appeared selected by default. This covers first launch, newly added cores, and runs from a directory without saved settings, preventing erroneous core updates until users explicitly opt in and save their selection.

kewalker, 2026-09-16:
Follow-up: the underlying updater path is now covered as well. Missing cores are initialized as disabled in SettingsService, so clicking Update immediately after a fresh install cannot treat every missing core as enabled. The PR still covers first launch, newly added cores, and directories without saved settings.

closed 2026-09-18 by RetroDriven