RetroDriven/Pocket_Updater #91: No Internet Connection Detected!

TheIcehawk, 2024-11-17:
Went to update my cores after a while but now getting this error whenever opening Pocket Updater.  Tried deleting and reinstalling the latest 1.6.1 release and still same issue, is this a dead project, there have been no updates since May?

Real-Dampfwalze, 2026-08-16:
Got this error message today. Maybe problems with a new Windows update? Or is the repository just down?

AnotherPocket, 2026-08-17:
I've been getting this error the last few days. Archive.org was down for a bit, but everything is back now. Also p_update is pulling assets without issue.

Real-Dampfwalze, 2026-08-18:
Ah, thank you. So the error message is a bit misleading. I thought the problem was on my computer.
I hope archive.org will never vanish. Without the pocket updater I would be helpless. :-)

Belzakare, 2026-08-28:
Mark me down for the same set of issues as Real-Dampfwalze. Same no internet, same Newtonsoft.... not sure what to do next here.

RetroDriven, 2026-08-31:
I'm looking into this thanks for reporting it. I'll let you know when I have it all figured out.

RetroDriven, 2026-08-31:
Please give this release a test and let me know how things go: https://github.com/RetroDriven/Pocket_Updater/releases/tag/v1.6.2

Real-Dampfwalze, 2026-08-31:
> Please give this release a test and let me know how things go: https://github.com/RetroDriven/Pocket_Updater/releases/tag/v1.6.2

Works perfect for me. No more error messages. Started an update  and it cleared fine, updating everything without further problems.
Thank you very very much. Now I have my prefered Pocket tool back. 👍❤😉

closed 2026-09-16 by RetroDriven